jtaczanowski/ThinkPad-X201-Arrandale-macOS-OpenCore #1: Big Sur install

jamesfawcett, 2021-07-10:
Hi @jtaczanowski  I'm following these steps today, so I hope you don't mind me documenting my progress here to try convert my X201 from Clover + Catalina, to OpenCore + Big Sur!

Firstly I'm trying to get OpenCore to boot without UEFI support.

I am installing the dependencies from OpenCorePkg to create a legacy booting USB now! :)

jtaczanowski, 2021-07-10:
Hi @jamesfawcett good luck, I keep my fingers crossed! :)

jtaczanowski, 2021-07-10:
Just updated readme with info on generating PlatformInfo items - it should be done for MacBookPro11,1 :)

jamesfawcett, 2021-07-10:
Cool thank you! I'll get that swapped. It's still downloading the build_duet packages, its taking forever!

jtaczanowski, 2021-07-10:
I think you can just download latest release of OpenCore https://github.com/acidanthera/OpenCorePkg/releases/download/0.7.1/OpenCore-0.7.1-RELEASE.zip without building it yourself :)

jamesfawcett, 2021-07-10:
Oh that's great! Your EFI is running the existing Catalina install very well! I am going to take a disk backup, then attempt upgrade :) Thanks for your help1

jamesfawcett, 2021-07-10:
Big Sur install was seamless with your EFI, thanks very much for your help! :) 

It's strange that it will never power off, and the no keyboard at the OpenCore screen is a bit odd too. I'll have a look for solutions! :)

jtaczanowski, 2021-07-10:
@jamesfawcett  I'm glad you managed to get started BigSur on your X201 :) 
On my machine reboots and keyboard at the OpenCore screen is not working also - I don't know why.

Were you able to enable graphics acceleration? :)
In addition **second EFI directory** (after patched graphics kexts install) contains one small - but very important change.

```SystemProductName``` is changed to ```MacBookPro6,2``` - this change enabling native power management through appleLPC. You can check if your CPU is running with full frequency range by running:
```sudo powermeirics```
My output looks as follows:
```
**** Processor usage ****

System Average frequency as fraction of nominal: 105.76% (2813.33 Mhz)
Package 0 C-state residency: 19.20% (C2: 0.00% C3: 19.20% C6: 0.00% C7: 0.00% )
```
CPU frequency is scalling from 1,2GHz to over 3GHz :)

jamesfawcett, 2021-07-10:
That's a nice little tip thank you for that! Graphic Acceleration is working, but, I am still on 11,1!

`**** Processor usage ****

System Average frequency as fraction of nominal: 45.00% (1197.00 Mhz)
Package 0 C-state residency: 2.70% (C2: 0.00% C3: 2.70% C6: 0.00% C7: 0.00% )

Core 0 C-state residency: 18.46% (C3: 18.46% C6: 0.00% C7: 0.00% )`

I am going to try change to 6,2 now! I am excited for this!  It's quick now!

Thanks
James

jamesfawcett, 2021-07-10:
Wow! That's impressive!

As a MacBookPro6,2 and the SSD, this machine is lightning quick! 

```
**** Processor usage ****

System Average frequency as fraction of nominal: 109.86% (2922.28 Mhz)
Package 0 C-state residency: 0.00% (C2: 0.00% C3: 0.00% C6: 0.00% C7: 0.00% )
```

Thanks very much for all your hard work!

jamesfawcett, 2021-07-10:
I've done some digging into the keyboard not working at the picker too, there seems to still be an open issue, I think this could be related but I can't find a workaround. I tried an old Ps2KeyboardDxe.efi from 0.6.3 but it's the same.

https://github.com/acidanthera/bugtracker/issues/1236

jtaczanowski, 2021-07-10:
Nice to hear that everything worked for you :) I was also surprised how well this machine works. Since now, I have not found a single task that he cannot handle resonable fast and efficiently :)

The hardest problem and the most time consuming was to uderstand how to patch DSDT file and how appleLPC relies on it, but i couldn't leave i7 in my thinkpad running at 1.2GHz :D

I'm looking for the issue with keyboard:
> For the record, current version (0.6.3) works with Ps2KeyboardDxe.efi from 0.6.2. Did not checked mouse.

Maybe it worth to try with Ps2KeyboardDxe.efi from 0.6.2 **NOT** from 0.6.3?

jamesfawcett, 2021-07-11:
Just tried - same results unfortunately, but was worth a try! Tried with each plugin of VooDooPs2Controller enabled/disabled too but I think they are mainly for within Mac OS not for the boot screen.

I'll keep looking :) I just need to be able to boot Windows 7 on this x201 because I use it for car diagnostics too, I can always boot up a Clover USB stick and select windows 7 though until I find a solution 

Thanks
James

jamesfawcett, 2021-07-12:
Interestingly I found something out: 
I enabled OpenCanopy and a GUI picker (set to external).
When it first loads, you can use the trackpad to select an OS. However, if you press any of the keyboard keys, then the mouse stops responding completely. 

Next step I'm going to compare to my X2100 config to see if anything missed, the internal picker works great for me withj this setup, and its the same keyboard :) https://github.com/jamesfawcett/Thinkpad-X2100-51nb-OpenCore-Hackintosh/blob/main/EFI/OC/config.plist

nijhawank, 2021-07-24:
This is awesome community effort. I have a perfectly running (minus sleep) T410i with Mojave with Clover, I’m gonna try this config to adapt for it, plus the Big Sur is a great news

ahmedsayed-0, 2021-08-10:
A workaround for the keyboard not working is to add AptioInputFix.efi in the Drivers folder
[AptioInputFix.zip](https://github.com/jtaczanowski/ThinkPad-X201-Arrandale-macOS-OpenCore/files/6959989/AptioInputFix.zip) this is the one i used to get the keyboard working

jamesfawcett, 2021-08-10:
> A workaround for the keyboard not working is to add AptioInputFix.efi in the Drivers folder
> [AptioInputFix.zip](https://github.com/jtaczanowski/ThinkPad-X201-Arrandale-macOS-OpenCore/files/6959989/AptioInputFix.zip) this is the one I used

Thanks for the suggestion, unfortunately this did not work. Still the same, no keyboard input. Was worth a try though, thanks!

ahmedsayed-0, 2021-08-10:
I'm using a different old laptop but I had the same keyboard problem, it worked after adding it and then doing a snapshot by opentree, I hope you find something that works for you.