jtaczanowski/ThinkPad-X201-Arrandale-macOS-OpenCore #2: Issue

Odyssey346, 2021-08-11:
Hi, I've installed Big Sur on my X201 through this guide. After rebooting from the installer, my screen is a blinking cursor. What do I do to fix this? Do I have to wait? Because on my UEFI patched OpenCore install USB it was lightning fast to show something, so I wouldn't think it'd be slower on a sata ssd.

jtaczanowski, 2021-08-11:
Hi :)
Did you install OpenCore on your sata SSD? To boot from sata SSD you have to install OpenCore on it after BigSur instalation process. So, for the first system boot - after installation, you should use OpenCore from your USB drive, boot BigSur and install OpenCore on your SATA ssd.

Odyssey346, 2021-08-11:
Is it an option in the OpenCore bootloader or something? I don't really understand

Odyssey346, 2021-08-11:
Ok, I read a bit, I'll try again later

jtaczanowski, 2021-08-11:
Install OpenCore on sata ssd in the same way like on your installation pendrive - this is exactly the same procedure.

Odyssey346, 2021-08-12:
Hmm, I did do that but, well, OpenCore legacy boot patcher doesn't work in macOS recovery. Do I just copy all the files from my USB over to the efi partition?

Odyssey346, 2021-08-12:
I see that I can go into macOS installer from the opencore usb i have, but my usb keyboard wont work in it too. I guess I can try another USB keyboard I have...

Odyssey346, 2021-08-12:
Great, I found a USB keyboard that works :) Too bad it's my dad's keyboard...

Odyssey346, 2021-08-12:
Ok, everything works now :D

closed 2021-08-12 by Odyssey346

jtaczanowski, 2021-08-12:
@Odyssey346 nice to hear that everything works :) Did you change SystemProductName to MacBookPro6,2 to achieve full CPU speeds?

Odyssey346, 2021-08-12:
Yeah, I did.