jtaczanowski/ThinkPad-X201-Arrandale-macOS-OpenCore #4: A few questions

makintos90, 2021-08-14:
First of all, thanks a lot for the work on this hack, its really useful!
Questions
1. Why haven't you used SSDT-HPET?
2. Why have you incuded DSDT.aml? The dortania guide mentions not to.
3. How did you get graphics working without an eGPU? The dortania guide only mentions support for upto 10.13.6 for integrated graphics.
Again, thanks a lot

jtaczanowski, 2021-08-15:
Hi @makintos90 :)

1) I did not playing with SSDT-HPET patches because I did not notice problems with my x201 that indicate interrupt conflicts such as not working devices. I only include prebuilt SSDTs mentioned in https://dortania.github.io/Getting-Started-With-ACPI/ssdt-methods/ssdt-prebuilt.html#laptop-clarksfield-and-arrandale. In free time I will check if running SSDTTime https://dortania.github.io/Getting-Started-With-ACPI/ssdt-methods/ssdt-easy.html#running-ssdttime will show something interesting.

2) I included DSDT.aml which is originally taken from https://github.com/jamesfawcett/Thinkpad-X201-Hackintosh-Catalina/blob/master/EFI/EFI/CLOVER/ACPI/patched/DSDT.aml which probably includes patches for battery indicator. From what I remember, running BigSur without this DSDT.aml resulted no battery indicator in menu bar. I know that including DSDT.aml especially not from your own machine - is not recommended, but it just works ;)
Another reason is that I have to patch LPC device-id from `pci8086,3b07` to `pci8086,3b09` which described in readme https://github.com/jtaczanowski/ThinkPad-X201-Arrandale-macOS-OpenCore#getting-cpu-power-management-intel-speedstep-to-work.
  I opened previously mentioned DSDT.aml in MaciASL and patched it with modified version of this file [misc_Haswell-LPC.txt](https://github.com/RehabMan/Laptop-DSDT-Patch/blob/master/misc/misc_Haswell-LPC.txt)
  ```
  #Maintained by: RehabMan for: Laptop Patches
  #misc_Haswell-LPC.txt
  
  # inject "compatible" with recognized series-8 LPC device-id
  into method label _DSM parent_adr 0x001F0000 remove_entry;
  into device name_adr 0x001F0000 insert
  begin
  Method (_DSM, 4, NotSerialized)\n
  {\n
      If (LEqual (Arg2, Zero)) { Return (Buffer() { 0x03 } ) }\n
      Return (Package()\n
      {\n
          "compatible", "pci8086,3b09",\n
      })\n
  }\n
  end;
  ```
 I saved patched DSDT.aml and copy to my EFI directory - patching was successful, because appleLPC start working.
 I'm not ACPI patching expert, maybe instead of including full DSDT.aml file witch patches I only need to generate this SSDTs files contains specific patches for battery indicator and LPC device-id?

3) I noticed that OpenCore Legacy Patcher (made for older MacBooks) includes @ASentientBot experimental patched kext and frameworks for older graphics cards which are described [here](https://dortania.github.io/OpenCore-Legacy-Patcher/ACCEL.html#downloading-older-non-metal-apps) and [here](https://github.com/dortania/OpenCore-Legacy-Patcher/issues/108). I decided to try running this graphics patches on x201 which has hw spec similar to MacBookPro6,2 and that was the trick - it works great in the same way like on original MacBookPro6,2.
I don't know why it is dedicated only for MacBooks and not included in OpenCore (only in OpenCore Legacy Patcher)

Regards

jtaczanowski, 2021-08-15:
Now I found some inconsistency - here Dortania guide mentions about patching and including DSDT.aml file https://dortania.github.io/OpenCore-Post-Install/laptop-specific/battery.html#dsdt-patching
It is not clearly for me how differs DSDT patching from SSDT patches.

jamesfawcett, 2021-08-16:
I'm enjoying following this!

In the linked DSDT above, I followed a guide to apply patches to the original DSDT from the i7-x201. It was during this commit here, that I swapped from separate SSDT's, to one DSDT. https://github.com/jamesfawcett/Thinkpad-X201-Hackintosh-Catalina/commit/27e62581b5ef6a894d9e84d136365ace934094a9#diff-c9ed26908c9907666f2bfa8d0ce5c310a6a1458a5286473e288058e8ce8b054f

I'm not sure if that is still the best method today, the original SSDT's came from the High Sierra guide here: https://osxlatitude.com/forums/topic/10136-short-guide-install-high-sierra-on-lenovo-x201-i5-520m/ (but were for an i5 model)

Looking forward to seeing where this project goes! Next stop - Monterey! :)  

My X201 i7 is working great on Opencore now (thank you!). I would love to get triple boot back, I'm still looking for solutions for the keyboard input on the picker.

jtaczanowski, 2021-08-17:
Hi @jamesfawcett :)
Thanks for information about DSDT file which helped me to launch macOS on my x201 :)
Maybe did you remember wich patches did you applied to DSDT? That would be valuable information.

Regarding to the Monterey, I think the biggest problem will be the graphics drivers. Already with BigSur it was difficult to revert back graphics acceleration on olds GPUs and it took ASentientBot several months  from the premiere of the system. I keep my fingers crossed for doing that again in Monterey.

Regarding to the not working keyboard on the picker - on your suggestion, I tried with enabled OpenCanopy. With that I could use trackpad, point on the entry wich I would to choose, and click it by tapping trackpad. It's not a perfect solution, but it working and I thought it would solve your problem :)

jtaczanowski, 2021-09-08:
Hi @jamesfawcett 

Good news :) https://forums.macrumors.com/threads/macos-12-monterey-on-unsupported-macs-thread.2299557/page-55?post=30238958#post-30238958
If Monterey can run on MBP6,2  there is a chance to run it on X201 :]

jamesfawcett, 2021-09-08:
[image]

That's great news! I am using mine daily at the moment (thank you!) I've taken my x2100 apart to swap from 2880x1920 to a 3000x2000 display... and you know what... there's nothing wrong with this x201! 

It's snappy, fast, does everything I need. I do wish it had a few more pixels but since I swapped it to the 1440x900 display from an x200 it's been great! I really do like this machine!

I've solved booting into windows 7 now. I still use the external picker with the mouse, I have to boot into the windows 10 boot manager then select Windows 7. For some reason, in clover I could boot straight to windows 7 (if i use the clover efi), but not OpenCore.

I appreciate you also having an interest in this! It's a fun project, and I love the fact my x201 only cost £100 and runs the latest MacOS!