jtaczanowski/ThinkPad-X201-Arrandale-macOS-OpenCore #16: Black screen after Post Install Root Patch

osmadro, 2022-07-07:
Hi :)

After downloading Open Core Legacy Patcher and follow [https://github.com/jtaczanowski/ThinkPad-X201-Arrandale-macOS-OpenCore#getting-graphics-qeci-acceleration-to-work](url) my T410 shows black screen after the verbose bootup. Any fix for that?

Grats,
Osmadro

jtaczanowski, 2022-07-08:
Hi @osmadro  :)
Did jutro try just reboot your computer? Similar situation happened to me one time - after installing Post Install Root Patch  I had black screen. After turning on and off my computer it started working.
What graphics do you have? Intel HD?

osmadro, 2022-07-08:
Yes sir, I rebooted but every time I have black screen. I swap the SSD to my father's X201 and its works,, but in my T410 not :(
I have Intel HD (Intel I5-540M) and without patching works, but really really slow. Every time I boot up the laptop load everything until the login, only the verbose is showing and then black screen. I remember one time after the verbose, the laptop showed a 🚫 logo with the verbose glitched. I used your EFI and I follow all the methods you used. Hope you can help me with this :)

jtaczanowski, 2022-07-08:
Hmm, I think that included ```DSDT.aml``` which is extracted from x201 may cause problems. Please try to disable it in config.plist https://github.com/jtaczanowski/ThinkPad-X201-Arrandale-macOS-OpenCore/blob/main/EFI/OC/config.plist#L20 (```<key>Disabled</key>```) and check if will this solve the problem.
```DSDT.aml``` included in this repo had applied patches for battery indicator and apple lpc device address (CPU power management). Maybe it will be needed to extract ```DSDT.aml``` from your t410 and apply patches again.

osmadro, 2022-07-08:
im going to try it, hope its work

osmadro, 2022-07-08:
nope, its not working, i edited the config.plist with propertree ¡, changin the DSDT.aml string to False and its not working. If you need files from my EFI of my t410 tell me and i will link it here. Btw im new at hackintosh so maybe I will have some difficulties with the files :)

osmadro, 2022-07-08:
well, thanks for you help :) I can confirm that the project work in x201. Sadly I erase the SSD and staring to do a hackintosh on another laptop I have with an Nvidia graphic card. I will close this. Maybe in the future I give another try. TY so much @jtaczanowski

closed 2022-07-09 by osmadro