jtaczanowski/ThinkPad-X201-Arrandale-macOS-OpenCore #17: Not booting

Rallo00, 2022-08-24:
Hi, I'm not new to hackintosh.
I've created the media with your EFI and BigSur installer, but when I try to boot from the pendrive the screen goes black with the underscore blinking.
I also updated the bios firmware to the latest 1.40, I've set the Intel Virtualization settings as you said in the BIOS.
I thought there was some problem with the PC, so I tried booting the Windows 7 Installer and works with no problem.

Have you got any idea why? At the beginning I thought maybe your EFI had UEFI boot mode but turns out it hasn't.

Please Help, thanks in advance.

jamesfawcett, 2022-08-24:
Hi @Rallo00 did you follow the steps for OpenCore Legacy booting?

Rallo00, 2024-01-02:
Hi, in fact I did. Unfortunately I do not have this laptop anymore, sorry for the late reply. I will close the issue as not planned. Thanks for your support anyway.

closed 2024-01-02 by Rallo00