jtaczanowski/ThinkPad-X201-Arrandale-macOS-OpenCore #21: WiFi won't stay connected (x201t)

ViktorNova, 2022-10-12:
Hello! Thanks for all of your work on this project. I installed onto an x201t with Big Sur, which went just fine (although install wouldn't complete with APFS, I had to use HFS+). 

I am booted into the new installation, and I can see wifi networks and connect to them, but it doesn't stay connected. It sometimes connects for a minute or two, and more often immediately disconnects. Sometimes it reconnects immediately, but usually it takes a few minutes of trying before it can reconnect. It's very random. 
I am right next to the access point, and while connected it shows full bars. 

I believe I'm using the stock wifi card that came with the laptop, but am not sure how to verify that. I followed your instructions exactly and am using your EFI as is, aside from swapping out the recommended things.
Wifi card is reporting as AirPort Extreme (0x8086, 0x1311) (I have not yet done the OpenCore legacy patcher, as I can't get it to download with intermittent connection, though not sure if that would help this or not)

Do you have any suggestions on how to make it more reliable?

ViktorNova, 2022-10-12:
@AndrzejL-eu I don't believe so. There is only one access point and I'm right next to it. And there are lots of other wifi devices in the house that have no issue. All of the errors in the kernel log make it seem like a driver issue, but I'm not really sure where to go next

jamesfawcett, 2022-10-12:
I swapped to a BroadCom AW-CB160 BCM94360HMB - it works out the box and has Airdrop etc.
I can't remember now if I needed Ipex-1 to ipex-4 adapters, would be worth checking. I also had a AW-CE123 which worked well too. That might have been the card that needed the adapters, check if you go down this route. Wifi signal is incredible with these cards too.

ViktorNova, 2022-10-12:
@jamesfawcett Thanks for the suggestion! Did you need to update BIOS or change/remove anything from the EFI included in this repo to get those working? I've seen old Clover guides for the x201 saying that you need to have a patched BIOS in order for it to work with a different wifi card but not sure if that applies to this method w/OpenCore and modern OS

jtaczanowski, 2022-10-13:
Hi @ViktorNova.
Could you try to connect to another AP, for example WiFi hotspot from telephone and check if problem also occurres?
I'm using stock WiFi card which is Intel 6300 thank's to https://github.com/OpenIntelWireless/itlwm. It mostly working fine except situations with multiple access points with similar signal strength which triggers roaming which is working not so reliable (As @AndrzejL-eu mentioned)
Airportitlwm is included in this repo for BigSur and Monterey.

Could you show Network->WIfI from your system report?

ViktorNova, 2022-10-18:
@jtaczanowski thanks for your response - I tried connecting to other APs and had the same issue, including a hotspot from my phone. 

I'm not 100% sure this is the stock wifi card as I got this thing used, but assuming it is. Also it's a X201T, which I wonder if it ships with a slightly different card than the standard X201? I'm happy to replace it if necessary but wanted to at least ask first:

Here is the output from system report: 
```
  Software Versions:
  CoreWLAN:	16.0 (1657)
  CoreWLANKit:	16.0 (1657)
  Menu Extra:	17.0 (1728)
  System Information:	15.0 (1502)
  IO80211 Family:	12.0 (1200.12.2b1)
  Diagnostics:	11.0 (1163)
  AirPort Utility:	6.3.9 (639.15)
  Interfaces:
en1:
  Card Type:	AirPort Extreme  (0x8086, 0x1311)
  Firmware Version:	itlwm: 2.1.0 fw: iwn-6000
  MAC Address:	18:3d:a2:82:c7:90
  Locale:	FCC
  Country Code:	US
  Supported PHY Modes:	802.11 a/b/g/n
  Supported Channels:	1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 36, 40, 44, 48, 52, 56, 60, 64, 100, 104, 108, 112, 116, 120, 124, 128, 132, 136, 140, 149, 153, 157, 161, 165
  Wake On Wireless:	Supported
  AirDrop:	Supported
  AirDrop Channel:	0
  Auto Unlock:	Supported
  Status:	Connected
  Current Network Information:
SkullParty:
  PHY Mode:	802.11n
  BSSID:	2c:7e:81:c6:85:da
  Channel:	11
  Country Code:	US
  Network Type:	Infrastructure
  Security:	WPA2 Personal
  Signal / Noise:	-56 dBm / -90 dBm
  Transmit Rate:	7
  MCS Index:	0
  Other Local Wi-Fi Networks:
Danger Alley:
  PHY Mode:	802.11
  BSSID:	02:83:cc:fe:d6:75
  Channel:	1,+1
  Country Code:	US
  Network Type:	Infrastructure
  Security:	WPA2 Personal
  Signal / Noise:	-74 dBm / -127 dBm
Danger Alley_EXT:
  PHY Mode:	802.11n
  BSSID:	10:0d:7f:96:5e:96
  Channel:	1,+1
  Network Type:	Infrastructure
  Security:	WPA2 Personal
  Signal / Noise:	-83 dBm / -127 dBm
bwanaspoons:
  PHY Mode:	802.11
  BSSID:	02:6a:e3:75:3e:bb
  Channel:	1,+1
  Country Code:	US
  Network Type:	Infrastructure
  Security:	WPA2 Personal
  Signal / Noise:	-64 dBm / -127 dBm
bwanaspoons2:
  PHY Mode:	802.11
  BSSID:	02:6a:e3:75:3e:bd
  Channel:	1,+1
  Country Code:	US
  Network Type:	Infrastructure
  Security:	WPA2 Personal
  Signal / Noise:	-63 dBm / -127 dBm

```

jtaczanowski, 2022-10-22:
I can see that you have stock Intel Wifi card which should work without any major problems.
Please show your `/var/log/wifi.log` file during problems occurs.

ViktorNova, 2022-10-24:
<details>
<summary>Click to expand..</summary>

Sun Oct 23 22:37:01.981 <airport[184]> ERROR: sharingd (415) is not entitled for com.apple.wifi.join_history, will not allow request
Sun Oct 23 22:37:01.981 <kernel> en1: Received EAPOL packet (length = 169)
Sun Oct 23 22:37:01.982 <kernel> inputEAPOLFrame: 0 extra bytes present in EAPOL frame.
Sun Oct 23 22:37:01.982 <kernel> inputEAPOLFrame: decrypting key data
Sun Oct 23 22:37:01.982 <kernel> inputEAPOLFrame: Received message 3 of 4
Sun Oct 23 22:37:01.982 <kernel> process3of4: Performing IE check.
Sun Oct 23 22:37:01.982 <kernel> process3of4: sending replyPacket ( len = 113 ).
Sun Oct 23 22:37:01.983 <kernel> process3of4: received pairwise GTK
Sun Oct 23 22:37:01.983 <kernel> ptkThread: Installing PTK!
Sun Oct 23 22:37:01.983 <kernel> PTK:
Sun Oct 23 22:37:01.983 [00000000] BF D3 6E 7B 8B 4B 9C 12 AD 87 70 D9 3E 9D DD 17
Sun Oct 23 22:37:01.983 <kernel> ptkThread: Installing GTK!
Sun Oct 23 22:37:01.983 <kernel> installGTK: setting cipher key (flags = 0x0)
Sun Oct 23 22:37:01.983 <kernel> GTK-RSC:
Sun Oct 23 22:37:01.983 [00000000] 07 80 16 00 00 00
Sun Oct 23 22:37:01.983 <kernel> GTK:
Sun Oct 23 22:37:01.983 [00000000] 69 AE 73 FB 21 24 EF F5 90 58 DE 00 04 1B 0C 33
Sun Oct 23 22:37:01.983 <kernel> installGTK: GTK installed
Sun Oct 23 22:37:01.995 <airport[184]> ERROR: rapportd (353) is not entitled for com.apple.wifi.join_history, will not allow request
Sun Oct 23 22:37:01.998 <airport[184]> ERROR: sharingd (415) is not entitled for com.apple.wifi.join_history, will not allow request
Sun Oct 23 22:37:02.027 <airport[184]> _processIPv4Changes: ARP/NDP offloads disabled, not programming the offload
Sun Oct 23 22:37:02.077 <airport[184]> _processIPv4Changes: ARP/NDP offloads disabled, not programming the offload
Sun Oct 23 22:37:02.170 <airport[184]> systemWokenByWiFi: System Wake Reason not found
Sun Oct 23 22:37:02.780 <airport[184]> _processIPv4Changes: ARP/NDP offloads disabled, not programming the offload
Sun Oct 23 22:37:03.074 <airport[184]> _processIPv4Changes: ARP/NDP offloads disabled, not programming the offload
Sun Oct 23 22:37:03.135 <airport[184]> systemWokenByWiFi: System Wake Reason not found
Sun Oct 23 22:37:11.639 <kernel> en1: link down event, informing supplicant.
Sun Oct 23 22:37:11.640 <kernel> postMessage::1412 APPLE80211_M_BSSID_CHANGED received
Sun Oct 23 22:37:11.640  Sun Oct 23 22:37:11.720 <kernel> AirportItlwm::getSSIDData(): Get failure: APPLE80211_IOC_SSID: -536870212
Sun Oct 23 22:37:11.882 <airport[184]> _processIPv4Changes: ARP/NDP offloads disabled, not programming the offload
Sun Oct 23 22:37:11.902 <airport[184]> _processIPv4Changes: ARP/NDP offloads disabled, not programming the offload
Sun Oct 23 22:37:13.587 <kernel> Unexpected payload found for message 9, dataLen 0
Sun Oct 23 22:37:13.588 <kernel> Error storing postMessage 9
Sun Oct 23 22:37:13.588 <kernel> postMessage::1412 APPLE80211_M_BSSID_CHANGED received
Sun Oct 23 22:37:13.588  <kernel> No InterfaceAverageCCA
Sun Oct 23 22:37:13.588 <kernel> IO80211ControllerMonitor::getChanCCA no fCCAReporterValue
Sun Oct 23 22:37:13.588 <kernel> Failed to getChanCCA e00002f0 36 0
Sun Oct 23 22:37:13.588 <kernel> No InterfaceAverageCCA
Sun Oct 23 22:37:13.588 <kernel> IO80211ControllerMonitor::getChanCCA no fCCAReporterValue
Sun Oct 23 22:37:13.588 <kernel> Failed to getChanCCA e00002f0 36 0
Sun Oct 23 22:37:13.601 <kernel> en1: Received EAPOL packet (length = 113)
Sun Oct 23 22:37:13.601 <kernel> inputEAPOLFrame: 0 extra bytes present in EAPOL frame.
Sun Oct 23 22:37:13.601 <kernel> inputEAPOLFrame: Received message 1 of 4
Sun Oct 23 22:37:13.601 <kernel> FULL RSN IE FOUND:
Sun Oct 23 22:37:13.601 [00000000] 30 14 01 00 00 0F AC 04 01 00 00 0F AC 04 01 00 00 0F AC 02 00 00
Sun Oct 23 22:37:13.601 <kernel> storeFullRSNIE: getAP_IE_LIST returned 0
Sun Oct 23 22:37:13.601 <kernel> parseRSNIE: groupCipherType = 5 pairwiseCipherType = 5 authSel = 2
Sun Oct 23 22:37:13.601 <kernel> process1of4: authsel is 2
Sun Oct 23 22:37:13.602 <kernel> PMK:
Sun Oct 23 22:37:13.602 [00000000] 55 7E 62 A1 74 76 14 A2 E8 17 C6 5C A9 56 BB A4 9B B3 8D B4 0F E3 3A 7B 43 79 6C 49 EA C5
Sun Oct 23 22:37:13.602 [0000001E] E0 04
Sun Oct 23 22:37:13.602 <kernel> TPTK:
Sun Oct 23 22:37:13.602 [00000000] 67 DB 77 54 F8 05 C4 F4 6C 8D 9A C7 E5 05 96 96 8B 10 6F 80 BD A5 1C 92 C3 F4 16 C8 B0 4B
Sun Oct 23 22:37:13.602 [0000001E] 6E 69 4E 3A 3F CA 79 B5 A7 5D 7C B6 CC 2B C9 30 C4 8B 00 00 00 00 00 00 00 00 00 00 00 00
Sun Oct 23 22:37:13.602 [0000003C] 00 00 00 00
Sun Oct 23 22:37:13.602 <kernel> KEY MIC:
Sun Oct 23 22:37:13.602 [00000000] 13 0B 71 AB AE 7D 04 90 14 45 F1 FC 8D 04 32 B6
Sun Oct 23 22:37:13.602 <kernel> process1of4: sending replyPacket 135 bytes
Sun Oct 23 22:37:13.604 <kernel> en1: link down event, informing supplicant.
Sun Oct 23 22:37:13.605 <kernel> postMessage::1412 APPLE80211_M_BSSID_CHANGED received
Sun Oct 23 22:37:13.605  Sun Oct 23 22:37:13.686 <kernel> AirportItlwm::getSSIDData(): Get failure: APPLE80211_IOC_SSID: -536870212
Sun Oct 23 22:37:13.689 <airport[184]> _handleLinkEvent: Unable to process link event, ssid request returned 5 (Input/output error)
Sun Oct 23 22:37:14.957 <airport[184]> systemWokenByWiFi: System Wake Reason not found
Sun Oct 23 22:37:19.788 <kernel> Unexpected payload found for message 9, dataLen 0
Sun Oct 23 22:37:19.789 <kernel> Error storing postMessage 9
Sun Oct 23 22:37:19.789 <kernel> postMessage::1412 APPLE80211_M_BSSID_CHANGED received
[... cut ...]
Sun Oct 23 22:37:51.826 [00000000] 65 CA BB 4B A4 70 C6 D1 72 59 A9 18 8C 5F 2F A8
Sun Oct 23 22:37:51.826 <kernel> process1of4: sending replyPacket 135 bytes
Sun Oct 23 22:37:51.826 <kernel> en1: link down event, informing supplicant.
Sun Oct 23 22:37:51.826 <kernel> postMessage::1412 APPLE80211_M_BSSID_CHANGED received
Sun Oct 23 22:37:51.826  <kernel> AirportItlwm::getSSIDData(): Get failure: APPLE80211_IOC_SSID: -536870212
Sun Oct 23 22:37:51.908 <airport[184]> _handleLinkEvent: Unable to process link event, ssid request returned 5 (Input/output error)
</details>

ViktorNova, 2022-10-31:
I am closing this issue, not because I got it working but because I just discovered that some hardware I need this Hackintosh for doesn't work on Big Sur, so I need to downgrade to Catalina. Thanks for all of your help though, it works great on an X-201t tablet!

closed 2022-10-31 by ViktorNova