jtaczanowski/ThinkPad-X201-Arrandale-macOS-OpenCore #23: Not booting OpenCore.

ters55, 2022-11-12:
Hello. I am trying to install Mac OS 11 Big Sur on a laptop with an Arrandale processor following your instructions. I moved your EFI folder, inside which the BOOT and OC folders are located, to the EFI partition on the flash drive, but the OpenCore bootloader does not start. What can be done?

ters55, 2022-11-12:
.

ters55, 2022-11-12:
.

ters55, 2022-11-12:
[image]

jtaczanowski, 2022-11-14:
Hi,
Did you follow instructions for legacy setup (for computers which not support UEFI, only MBR)? https://dortania.github.io/OpenCore-Install-Guide/installer-guide/mac-install.html#legacy-setup