jtaczanowski/ThinkPad-X201-Arrandale-macOS-OpenCore #24: Trouble with HfsPlusLegacy.efi

fajnykod, 2023-01-03:
Hi,

I'm trying to install this on a Lenovo Thinkpad X201 i5-540m, I have done this in the past and it had worked but now when I make a USB the same way as in the guide it won't boot into the installer, says something like HfsPlusLegacy.efi at 3 - unsupported! I have tried to just skip the file by making it false in the config.plist and it boots but just hangs at some text with # at the start. What's going on here and how can I fix this? 

(I am using the latest OpenCore release 0.8.8-RELEASE and a cloned version of this repo if that helps.)

Thanks! (Log here: [opencore-2023-01-03-165520.txt](https://github.com/jtaczanowski/ThinkPad-X201-Arrandale-macOS-OpenCore/files/10338398/opencore-2023-01-03-165520.txt))

Scorpionchiques, 2023-01-03:
> Hi,
> 
> I'm trying to install this on a Lenovo Thinkpad X201 i5-540m, I have done this in the past and it had worked but now when I make a USB the same way as in the guide it won't boot into the installer, says something like HfsPlusLegacy.efi at 3 - unsupported! I have tried to just skip the file by making it false in the config.plist and it boots but just hangs at some text with # at the start. What's going on here and how can I fix this?
> 
> (I am using the latest OpenCore release 0.8.8-RELEASE and a cloned version of this repo if that helps.)
> 
> Thanks! (Log here: [opencore-2023-01-03-160918.txt](https://github.com/jtaczanowski/ThinkPad-X201-Arrandale-macOS-OpenCore/files/10338445/opencore-2023-01-03-160918.txt))

You shuld use opencore from this repo 0.8.0

fajnykod, 2023-01-05:
Thanks! I'll test it out once I install the tools to make the OpenCore USB again.