jtaczanowski/ThinkPad-X201-Arrandale-macOS-OpenCore #25: Problem with Post Install - several issues

BT-Lim2021, 2023-05-08:
Hi jtaczanowski, Thank You very much for this.
I am very much a newbie with github as well as hackintosh.  Just started learning to understand them this weekend.
In anycase, after much confusion, I manage to successfully create the USB Installer.
And it worked.  I have successfully installed Big Sur on my Thinkpad x201 (i5)..

But then the problem began.  I was not able to boot from the internal harddrive.  I needed to boot from the USB drive.
That's when I realized I needed to do the Post install steps.

**MAIN ISSUES:**

1. A Bigger problem arise.  There is no EFI partition associated with the main drive.  
2. and when I tried to execute "BootInstall_X64.tool"  it gives me an error "No FAT32 partition to install"

What did I do wrong?  Did I missed a step at the beginning?
Any suggestions on what I need to do?  Please?  and Thank You in advance for your help and support.

P.S. I noticed during the creation of the USB drive, when I ran the "BootInstall_X64.tool", it did not asked me if I want to create a new MBR.  I am not sure if the updated script has done away with it or it was a problem for me.

So, that is the biggest problem I faced so far.

**SMALLER ISSUE:**
Another smaller issue but important to me:  I have the docking station, and I connect a monitor through display port on the docking station.  Big Sur does not recognize that there is a monitor connected.

jtaczanowski, 2023-05-10:
Hi @BT-Lim2021 
I'm glad to know that you was able to install BigSur on your Thinkpad.

I haven't tried to get display port to work. 
I think the problem is related to this part of configuration https://github.com/jtaczanowski/ThinkPad-X201-Arrandale-macOS-OpenCore/blob/main/EFI/OC/config.plist#L373:L384
and lack of entries about external port, something like that:
```
framebuffer-con1-enable = 01000000
framebuffer-con1-type = 04000000 (or 00020000)
```
Information about that configuration are here https://dortania.github.io/OpenCore-Post-Install/gpu-patching/intel-patching/connector.html#patching-connector-types
You can try add this entries to config.plist with this tool [ProperTree](https://github.com/corpnewt/ProperTree) and check if it will be working :)
In free time I will also check that.

Regards

BT-Lim2021, 2023-05-14:
Hi @jtaczanowski,
Thank You for the suggestion above and apologies took me a few days to give an update.  Long story but need to restart the whole installation process.
I have bad news and good news update.

The bad news:  Tried the above suggestions and amended entries.  No Joy unfortunately.
My PC only show 1 display.  So, there is no conX.
I have tried different combinations, eg:  framebuffer-con0-type, framebuffer-con-type; and with values 04000000 or 00020000.
None of them works.
What that means is that I cannot use both the thinkpad LCD Display and external monitor through Display Port on the ultrabase.

The Good news:
I changed the BIOS:  Config > Display to Digital on ThinkPad (from ThinkPad LCD).
The DisplayPort works. Obviously nothing on the ThinkPad LCD (blank/ inactive).  Don't need to do anything different in config.plist.
It also worked if I reboot with the monitor closed, as it defaults to attached DP monitor. So, you probably don't need to even change the Bios Display setting.
This works for me because there are some damage on my thinkpad LCD screen (blury) so I don't intend to use it.
Kinda kicking myself for not trying this earlier, seems like an obvious thing to try and do.

I faced one problem though, after I Install Root Patch with OpenCoreLegacy Root patch v0.4.1 the DP Display glitch out in the middle of mac booting up.  It is displaying properly at the beginning and then went blank and the monitor gives our "No Signal".  I even went back to put the old working EFI back, but still same problem.  Not sure if the Root Patch did something else.

I am trying to solve another challenge now: Dual Boot with Windows 11.
My current working set up:
1. Windows 11 installed on Thinkpad internal HDD (this is also hacked version to bypass the TPM 2.0 and Secure Boot requirments).  So it is installed on MBR now, not GPT.
2. MacOS Big Sur is installed on an external HDD.
3. This set up works but it means that I need to plug in the external HDD to boot MacOS, and Unplug to Boot to Windows 11 (with USB HDD set as the highest priority in BIOS Boot Setting)

I am trying to find a way to be able to select dual boot without needing to plug and unplug the USB HDD.
So far, not successful.  Tried two or three methods that I found through research, none of them work.  Even tried using eacyBCD but cannot detect my external HDD.  

Not sure if you have any idea, how to enable this dual boot.