jtaczanowski/ThinkPad-X201-Arrandale-macOS-OpenCore #27: graphics QE/CI acceleration not working

rivciks, 2023-07-06:
Hallo,
I have a Thinkpad t410 but my graphics card is the same as in your model: Intel HD Graphics 288 MB.

I did everything as described and also ran OpenCore-Patcher and applied the Post Install Root Patch. The system was patch, as I am able to tell,  without errors but it seems that the  QE/CI acceleration is not working.  Everything runs really slow and the VDADecoderCheck also fails.

[image]

jtaczanowski, 2023-07-16:
Hi :)
Have you checked if this issue existing also in Monterey installation?
Maybe there is differences between x201 and t410. From what I remember I have made some changes in this section of `config.plist` file:
https://github.com/jtaczanowski/ThinkPad-X201-Arrandale-macOS-OpenCore/blob/main/EFI/OC/config.plist#L361:L388
to get iGPU working correctly with post install patches.
One of instructions on which I based was this manual: https://github.com/acidanthera/WhateverGreen/blob/master/Manual/FAQ.IntelHD.en.md

rivciks, 2023-07-18:
Thanx,  
After further examination it seems that QE/CI acceleration is actually working despite failed DADecoderCheck.

closed 2023-07-18 by rivciks