jtaczanowski/ThinkPad-X201-Arrandale-macOS-OpenCore #28: CPU Power Management with Speedstep

rivciks, 2023-07-06:
Hello again, 
I have also a question: how do you enable the Intel CPU speedstep on the Thinkpad? Do you have to download a special tool or something like that?

P.S. Thanx for the guide. 👍

jtaczanowski, 2023-07-16:
To enable speedstep I have to patch/spoof LPC device-id `pci8086,3b07` -> `pci8086,3b09` to match LPC device-id from real MacBookPro6,2 `https://github.com/jtaczanowski/ThinkPad-X201-Arrandale-macOS-OpenCore#information-about-cpu-power-management-intel-speedstep
I used MaciASL tool to patch this LPC device-id.
More detailed informations about patching process I described here:
https://github.com/jtaczanowski/ThinkPad-X201-Arrandale-macOS-OpenCore/issues/4#issuecomment-899125408

rivciks, 2023-07-18:
OK. Thank You.

closed 2023-07-18 by rivciks