jtaczanowski/ThinkPad-X201-Arrandale-macOS-OpenCore #29: IOConsoleUsers: gIOScreenLockState 3 Error after installing drivers with OCLP

Sarpmanon, 2023-10-02:
Hello @jtaczanowski, I don’t know if you’re still using GitHub or mess with Hackintosh but I have an Arrandale laptop and I followed your guide to install Big Sur. Before installing the drivers, my pc was booting perfectly and had no issues booting or running (except hardware acceleration). After preparing the USB, I ran OCLP and after restarting I started to get this error: IOConsoleUsers: gIOScreenLockState 3, hs 0, bs 0, now 0, sn 0x0.

Thank you for reading , have a nice day 👋

jtaczanowski, 2023-10-07:
Hi :)
What exact laptop model do you have?
I think that probably you have to set different settings in this section of `config.plist`
https://github.com/jtaczanowski/ThinkPad-X201-Arrandale-macOS-OpenCore/blob/main/EFI/OC/config.plist#L364:L384

To get GPU configuration to get work with acceleration I used below documentations.
https://github.com/acidanthera/WhateverGreen/blob/master/Manual/FAQ.IntelHD.en.md#intel-hd-graphics-first-generation--ironlake-arrandale-processors
https://dortania.github.io/OpenCore-Post-Install/gpu-patching/intel-patching

Unfortunately patching iGPU to get it work was hard. I went through many iterations of reading and  trying to understand how it is working, changing configuration, checking if it is working and over and over again.

Thank you, best regards too :)

Sarpmanon, 2023-10-07:
Hello, i have a Sony Vaio VPCEB33FX to be exact and i'm using my own config that was working before the OCLP patches. Unfortunately GPU patching for my Arrandale isnt working and Hackintosh doesnt boot when i try to use the guides you've sent.

Btw, i'm trying to use Big Sur

Thank you for your kind reply, cheers :)

soyeonswife63, 2024-03-28:
> Hello @jtaczanowski, I don’t know if you’re still using GitHub or mess with Hackintosh but I have an Arrandale laptop and I followed your guide to install Big Sur. Before installing the drivers, my [...]

had this exact issue! i solved it by:
- copying iGPU DeviceProperties from this EFI
- i was on MacBookPro11,1 for a bit for the install, changed it to MacBookPro6,2
- copied over kernel patches from this EFI

and with that i was up and running! good luck :)