jtaczanowski/ThinkPad-X201-Arrandale-macOS-OpenCore #30: Problem with wifi in the installer

KorneliuszTryzna, 2023-10-29:
Hey @jtaczanowski ! I was wondering if you could help me with the installation of the macos. My issue is, the wifi disconnects when the "continue" button is pressed. Due to that, the installation does not want to continue. My wifi card is Intel Centrino Advanced N-6200. The thinkpad is the x201 model but with the i5-520m. I would really appreciate if you could help me, because in the near future it will be a whole month of trying! Thanks in advance.

jtaczanowski, 2023-11-01:
Please check if offline installer is working fine. In case of using offline installer it should working even without working WiFi connection.

jtaczanowski, 2023-11-06:
I received information from a @KorneliuszTryzna that the offline installer helped.

jtaczanowski, 2023-11-14:
Issue with online installer is probably related with this:
https://github.com/jtaczanowski/ThinkPad-X201-Arrandale-macOS-OpenCore/issues/19#issuecomment-1562414630