jtaczanowski/ThinkPad-X201-Arrandale-macOS-OpenCore #31: macOS Monterey needs BlueToolFixup

mhmtnasir, 2024-03-13:
Hello! BlueToolFixup is required on macOS Monterey and newer for Bluetooth to work. This applies to USB cards as well.
Adding BlueToolFixup into EFI would probably fix your BT issue! You might consider doing this.

jtaczanowski, 2024-03-18:
Hello,
In the past I tried to add BlueToolFixup but unfortunately it did not help. In free time I will try again.
Anyway, thank you for the tip! :)

mhmtnasir, 2024-03-29:
> Hello, In the past I tried to add BlueToolFixup but unfortunately it did not help. In free time I will try again. Anyway, thank you for the tip! :)

I'm guessing it was added along with the other kexts? (BlueToolFixup, BrcmBluetoothInjector, BrcmFirmwareData and BrcmPatchRAM3)
That's weird, the Cambridge chip should be recognized as a Broadcom card and supported with those kexts. I currently have the above 4 kexts on L512 (also Arrandale!) and Bluetooth works in Monterey.