jtaczanowski/ThinkPad-X201-Arrandale-macOS-OpenCore #32: I can't use your efi, please help

zerooor, 2024-05-22:
I have similar laptop from another firm: Asus k52f. also uses Intel arrandale (i5-560m) help me install hack

jtaczanowski, 2024-05-24:
Which step is unclear or what problem did you encounter?

zerooor, 2024-05-24:
Use debug version of opencore because I'm not professional this case)

jtaczanowski, 2024-05-27:
Does the OpenCore menu appear? If it does, what boot entries are on the list?

zerooor, 2024-05-27:
[image]
[image]
[image]

jtaczanowski, 2024-05-29:
I'm not sure - but it looks like there is no macOS installer entry (`Install macOS Monterey (external)`) in the menu.
Did you setup macOS installer on your external drive before setting up OpenCore bootloader on that drive?

zerooor, 2024-05-29:
I used the macrecovery script

jtaczanowski, 2024-05-29:
Please try with offline installer - I've never tried install from online installers. Another user reported that online installer was not working (https://github.com/jtaczanowski/ThinkPad-X201-Arrandale-macOS-OpenCore/issues/19#issuecomment-1562414630) without changes in `config.plist`

zerooor, 2024-05-29:
And how do I make offline from linux?

zerooor, 2024-06-03:
Hello Jan. I created an offline image from under a macos virtual machine, but now another problem has appeared. After choosing install macOS High Sierra, my laptop just reboots after a few seconds

jtaczanowski, 2024-06-05:
Hello @zerooor, EFI configuration is excepted to work with macOS BigSur or Monterey.
Could you check with one of this macOS versions?