jtaczanowski/ThinkPad-X201-Arrandale-macOS-OpenCore #33: Automatically shut down the during installation

t-vm, 2024-07-25:
Hi, I used your solution to install macOS11.7, but encountered problems:-(
- After selecting the boot disk, there will be a warning line: OCS: No schema for Block at 0 index, context <Emulate>! It does not seem to affect the subsequent startup, so I ignored it
- Then I installed the system according to the normal steps, and the computer tried to restart when the progress bar reached about three-quarters.
- After restart, I selected macOS installer, but from this time on, the installation progress will start from "29 mins remaining" after each restart, and the computer will automatically shut down when the progress bar reaches about one-third.
- No matter how many times I repeat, the installation progress will not make any more  progress.
- After I repeated the restart-installation-shutdown process about four or five times, when I tried to enter the macOS installer or Install macOS BigSur from oc again, the computer would automatically shut down directly after some codes appeared.

In addition, I think this may be related to the version of the system image or `bootX64`, but I am not sure. Can you please tell me the version of the system image and opencore you are using?
plz help me!

jamesfawcett, 2024-07-26:
Hi @t-vm please can you enable verbose (add -v to your boot-args) and let us know if there's a message showing when it gets stuck? The repo version is using OC 0.8.0

t-vm, 2024-07-26:
> Hi @t-vm please can you enable verbose (add -v to your boot-args) and let us know if there's a message showing when it gets stuck? The repo version is using OC 0.8.0

thanks for your reply!!  I will try that and tell you the result.

t-vm, 2024-07-26:
> Hi @t-vm please can you enable verbose (add -v to your boot-args) and let us know if there's a message showing when it gets stuck? The repo version is using OC 0.8.0

Hi @jamesfawcett , I tried many times again, still the same problems. I will provide some more detailed information:
What I have tried:
- Replace the new USB drive as the installation medium
- Update the information in step 2
- Use `bootX64` in OC version 0.8.0
- Use a new system image
The above attempts did not work at all.
My device is X201i, 8GB RAM (1600Mhz) and i5 450m. 
Now, in addition to the previous problems, my computer has a chance to shut down automatically before entering OC (before that I selected the boot item, the warning `OCS: No schema for Block at 0 index, context <Emulate>! ` was then displayed, and then the computer shut down)

In addition, a detail is that sometimes the computer will beep when I select the boot item or shut down, I don’t know if this is useful information

The following is the log generated in the `EFI` volume:
```
00:000 00:000 OCS: No schema for Block at 0 index, context <Emulate>!
15:739 15:739 LNX: AutodetectLinux not root fs - Not Found
16:844 01:104 AAPL: #[EB.H.IS|!] Err(0xE) <- RT.GV boot-signature 7C436110-AB2A-4BBB-A880-FE41995C9F82
16:871 00:027 AAPL: #[EB.H.IS|!] Err(0xE) <- RT.GV boot-image-key 7C436110-AB2A-4BBB-A880-FE41995C9F82
16:898 00:026 AAPL: #[EB|H:IS] 0
16:925 00:027 AAPL: #[EB|LOG:INIT] 2024-07-26T19:50:40
16:953 00:027 AAPL: #[EB|VERSION] <"bootbase.efi 495.140.2~3 (Official), built 2021-06-21T19:20:21-0700">
16:980 00:026 AAPL: #[EB|BUILD] <"BUILD-INFO[306]:{"DisplayName":"bootbase.efi","DisplayVersion":"495.140.2~3","RecordUuid":"D1225010-D2B5-46DE-95DA-C2176313E127","BuildTime":"2021-06-21T19:20:21-0700","ProjectName":"efiboot","ProductName":"bootbase.efi","SourceVersion":"495.140.2","BuildVersion":"3","BuildConfiguration":"Release","BuildType":"Official"}">
17:007 00:027 AAPL: #[EB.CFG.DEV|!] Err(0xE) <- RT.GV booter-strict-xmlparser 7C436110-AB2A-4BBB-A880-FE41995C9F82
17:035 00:027 AAPL: #[EB|CFG:DEV] r5 0x0 0x0
17:064 00:029 AAPL: #[EB|H:IS] 0
17:092 00:027 AAPL: #[EB|WL:MODE] 0
17:118 00:026 AAPL: #[EB|CFG:ARG] boot-save-log 0x0000000000000002 (0x0000000000000002 < 0xFFFFFFFFFFFFFFFF) default
17:146 00:027 AAPL: #[EB|CFG:ARG] wake-save-log 0x0000000000000002 (0x0000000000000002 < 0x0000000000000002) default
17:173 00:027 AAPL: #[EB|CFG:ARG] console       0x0000000000000001 (0x0000000000000001 < 0x0000000000000001) default
17:200 00:026 AAPL: #[EB|CFG:ARG] serial        0x0000000000000001 (0x0000000000000001 < 0x0000000000000000) default
17:227 00:027 AAPL: #[EB|CFG:ARG] embed-log-dt  0x0000000000000000 (0x0000000000000000 < 0x0000000000000000) default
17:255 00:027 AAPL: #[EB|CFG:ARG] timestamps    0x0000000000000000 (0x0000000000000000 < 0xFFFFFFFFFFFFFFFF) default
17:282 00:027 AAPL: #[EB|CFG:ARG] log-level     0x0000000000000001 (0x0000000000000001 & 0x0000000000000021) default
17:309 00:026 AAPL: #[EB|CFG:ARG] breakpoint    0x0000000000000000 (0x0000000000000000 & 0x0000000000000000) default
17:336 00:027 AAPL: #[EB|CFG:ARG] kc-read-size  0x0000000000100000 (0x0000000000100000 < 0xFFFFFFFFFFFFFFFF) default
17:364 00:027 AAPL: #[EB|H:IS] 0
17:391 00:027 AAPL: #[EB|WL] 0 0 0x01 0x01   0 0x00
17:418 00:026 AAPL: #[EB|BRD:SMBIOS] Mac-F22586C8
17:445 00:027 AAPL: #[EB|WL] 0 0 0x01 0x01   2 0x00
17:473 00:027 AAPL: #[EB.BST.IDT|+]
17:500 00:026 AAPL: #[EB.BST.IDT|-]
17:527 00:027 AAPL: #[EB|WL] 0 0 0x01 0x01   3 0x00
17:554 00:027 AAPL: #[EB|WL] 0 0 0x01 0x01   4 0x00
17:582 00:027 AAPL: #[EB|BRD:SMBIOS] Mac-F22586C8
17:609 00:026 AAPL: #[EB|WL] 0 0 0x01 0x01   5 0x00
17:636 00:027 AAPL: #[EB.H.CHK|BM] 0x0000000000000000
17:664 00:027 AAPL: #[EB.H.LV|!] Err(0xE) <- RT.GV boot-signature 7C436110-AB2A-4BBB-A880-FE41995C9F82
17:691 00:027 AAPL: #[EB|WL] 0 0 0x01 0x01  23 0x0E
17:718 00:026 AAPL: #[EB.H.LV|!] Err(0xE) <- RT.GV boot-image-key 7C436110-AB2A-4BBB-A880-FE41995C9F82
17:745 00:027 AAPL: #[EB|WL] 0 0 0x01 0x01  24 0x0E
17:772 00:027 AAPL: #[EB.H.LV|!] Err(0xE) <- RT.GV boot-image 7C436110-AB2A-4BBB-A880-FE41995C9F82
17:799 00:026 AAPL: #[EB.H.LV|!] Err(0xE) <- RT.SV- boot-signature 7C436110-AB2A-4BBB-A880-FE41995C9F82
17:827 00:027 AAPL: #[EB.H.LV|!] Err(0xE) <- RT.SV- boot-image-key 7C436110-AB2A-4BBB-A880-FE41995C9F82
17:854 00:027 AAPL: #[EB.H.LV|!] Err(0xE) <- RT.SV- boot-image 7C436110-AB2A-4BBB-A880-FE41995C9F82
17:881 00:027 AAPL: #[EB|H:NOT]
17:908 00:026 AAPL: #[EB|SB:P] 0x0
17:936 00:027 AAPL: #[EB|LIMG:DP] Acpi(PNP0A03,0)/Pci(1F|2)/SATA(0,FFFF)/HD(Part2,Sig53E02864-7B45-4706-B02A-2931066BA85A)/VenMedia(BE74FCF7-0B7C-49F3-9147-01F4042E6842)
17:963 00:027 AAPL: #[EB|LIMG:FP] \C5526162-9387-4191-BA3B-CA155B8BCCF1\com.apple.installer\boot.efi
17:990 00:026 AAPL: #[EB|LIMG:OPT] 
18:017 00:027 AAPL: #[EB.OPT.LXF|F] <"\\C5526162-9387-4191-BA3B-CA155B8BCCF1\\com.apple.installer\\com.apple.Boot.plist">
18:045 00:027 AAPL: #[EB.LD.LF|IN] 0 1 <"\\C5526162-9387-4191-BA3B-CA155B8BCCF1\\com.apple.installer\\com.apple.Boot.plist"> <"0">
18:073 00:027 AAPL: #[EB|KF] <"-no_panic_dialog -rootdmg-ramdisk auth-root-dmg=file:///macOS%20Install%20Data/BaseSystem.dmg lca-boot-mode=autoinstall-msu">
18:099 00:026 AAPL: #[EB|MBA:CL] <"">
18:127 00:027 AAPL: #[EB|MBA:NV] <"-v -no_compat_check keepsyms=1 debug=0x100 alcid=13 amfi_get_out_of_my_way=1 ipc_control_port_options=0">
[... cut ...]
23:344 00:029 AAPL: #[EB|RH:MF] <"usr\\standalone\\OS.dmg.root_hash.x86legacyap.im4m">
23:373 00:028 AAPL: #[EB.LD.LF|IN] 0 1 <"usr\\standalone\\OS.dmg.root_hash"> <"0">
23:401 00:028 AAPL: #[EB.LD.OFS|OPEN!] Err(0xE) <"usr\\standalone\\OS.dmg.root_hash">
23:429 00:027 AAPL: #[EB.RH.LRH|P!] Err(0xE) <- EB.LD.LF
23:457 00:028 AAPL: #[EB.BST.FBS|!] Err(0xE) <- EB.RH.LRH
23:486 00:028 AAPL: #[EB.B.WFDW|!WF] Err(0xE) 0
23:514 00:028 AAPL: #[EB|LOG:DT] 2024-07-26T19:50:47
23:542 00:028 AAPL: #[EB|LOG:EXITBS:START] 2024-07-26T19:50:47
```

t-vm, 2024-07-27:
> Hi @t-vm please can you enable verbose (add -v to your boot-args) and let us know if there's a message showing when it gets stuck? The repo version is using OC 0.8.0

@jamesfawcett Thanks, I have solved the problem now, I think it has something to do with sleep. As long as I keep giving keyboard input during installation, the it will complete.
But now I have a new problem, my wireless card is not working, I can't use Bluetooth and Wi-Fi. Do you have any solution? My  card is  6300agn