jtaczanowski/ThinkPad-X201-Arrandale-macOS-OpenCore #34: About the evolution on X201

BlockHackTk, 2024-10-24:
Hi Jan,

I own a Thinkpad X201 and would like to give it a try. I've never owned a Mac. I would like to know what is the most recent version I can run on this PC. I've upgraded it to 8GB RAM with a 1TB SSD. I would also like to know if it runs as well as Linux or Windows on the X201.

Thanks in advance.

BlockHackTk, 2024-10-24:
I'd also like to know if it's possible to set up this hackintosh without owning a Mac?
(As can be deduced from my first message).
I can prepare the media either under Windows or on Linux mangaro. 
Is this possible?

jtaczanowski, 2024-12-19:
Hi @BlockHackTk, 
The most recent version which I was using on x201 was macOS Monterey.
It works really snappy, just like Windows or Linux.
About installing from online installer please read this post https://github.com/jtaczanowski/ThinkPad-X201-Arrandale-macOS-OpenCore/issues/19#issuecomment-1562414630
You only have to change `SystemProductName` from `MacBookPro11,1` to `MacBookPro11,2`