jtaczanowski/ThinkPad-X201-Arrandale-macOS-OpenCore #36: OCB: LoadImage Failed - Volume Corrupt error

Heftycub0411, 2025-01-01:
After I copied the efi into my drive with big sur installed i got this error 
[image]
 I replaced the hfspluslegacy.efi file in drivers folder and it loaded into boot picker, but after selecting the big sur instalation it showed this error 
[image]
Tried different drives, monterey, downloaded the bigsur file again but nothing worked. I tried building my own efi it works with high Sierra but I don't know how to get it to work with bug sur. Arrandale Intel core i3 m380 on Acer aspire 5733 laptop.

bengillitt, 2025-12-09:
Hi Hefty :)

I've got the same issue, curious whether you found a way to fix it or not?

330d, 2026-01-11:
hi @bengillitt, were you able to fix it?